partyrobotics/bartendro #158: Database Upload Not Working Properly

pmich, 2015-01-10:
Several weeks back I added many ingredients and drinks to my bot and downloaded bartendro.db to my DropBox account. I renamed the file to be bartendroDec2014.db. Today I tried uploading that database file from the options tab. After, the file is uploaded, I get a message saying that the new database file has been successfully uploaded. However when I go to the drinks page, the new drinks are not present. Also, when I go back to the 'upload database' button, it says that the current database file is bartendro.db instead of bartendroDec2014.db like it should be. I've tried 'save and reboot' from the options screen, and I've even tried naming the file back to bartendro.db and uploading that, but all the newly added stuff still doesn't show up. I could've sworn this worked before. Thoughts?

OnkelOlli, 2015-01-10:
After adding new drinks and ingredients, and saving every time, you need to
reboot via the options menu. Then, the additions are used correctly and
will show up in the menu, and only then you can export the new  database.
If you do not reboot, somehow the old settings will be exported.
Greetings, Olli

nighteagle1974, 2015-12-04:
Hi,

have somewhere an working rpi B+ image?
I think the Image is onyl for RPI B old Version and ist not booting for an B+ RPI.