partyrobotics/bartendro #164: Liquid level sensor triggered for one ingredient, breaks all dispensing.

pmich, 2017-05-02:
Garran: I am trying to use the liquid level sensors on my Bartendo, but I am running into an issue. If one ingredient runs out it won't let me make any of the drinks. It gives a message saying "Bartendro has hit a snag and can't make drinks right now." Do you know what is going on here? Thanks!

Pierre: Might be worth checking the calibration values and making sure they're sane. Also, make sure that the ingredient that's running out (pretend it's vodka) isn't used by every other drink in the menu.

Garran: Yeah, the values are good. I can look at the dispensers and it shows that only one ingredient is empty (pretend it is vodka). It removes all drinks which require vodka from the menu. When I try to dispense a drink which doesn't contain vodka it lets me adjust the drink and when I hit pour it flashes the "now serving" popup which is immediately replaced with a message "Bartendro is broken. Bartendro has hit a snag and can't make drinks right now. Sorry!" I have tried power cycling it, and that doesn't change anything.

RPI B v. 1
software version is 20140501

ggossage, 2017-05-02:
The ingredient which has run out does not appear in the shots menu, which makes sense. All other ingredients CAN be dispensed from the shots menu. It is only the mixed drinks which can't be dispensed.

ggossage, 2017-05-03:
Updated software to 20151103. Issue appears to behave exactly the same as before.

Debug log:

Bartendro start up sequence:
Opening /dev/ttyAMA0
Done.

Opening I2C bus to router
Done.
Discovering dispensers
port 0:
  E7 E7 E7  -- Found dispenser with pump id E7, index 1
port 1:
  33 33 33  -- Found dispenser with pump id 33, index 2
port 2:
  BE BE BE  -- Found dispenser with pump id BE, index 3
port 3:
  61 61 61  -- Found dispenser with pump id 61, index 4
port 4:
  87 87 87  -- Found dispenser with pump id 87, index 5
port 5:
  F1 F1 F1  -- Found dispenser with pump id F1, index 6
port 6:
  29 29 29  -- Found dispenser with pump id 29, index 7
port 7:
port 8:
port 9:
port 10:
port 11:
port 12:
port 13:
port 14:
Detected dispensers version 2. (Only checked first dispenser)
mixer.check_liquid_levels: check levels
dispenser 0 level: 71 (stored: 0)
dispenser 1 level: 150 (stored: 0)
dispenser 2 level: 123 (stored: 0)
dispenser 3 level: 157 (stored: 0)
dispenser 4 level: 152 (stored: 0)
dispenser 5 level: 139 (stored: 0)
dispenser 6 level: 114 (stored: 0)
Checking levels done. New state: 0
mixer.check_liquid_levels: check levels
dispenser 0 level: 72 (stored: 0)
dispenser 1 level: 149 (stored: 1)
dispenser 2 level: 122 (stored: 1)
dispenser 3 level: 156 (stored: 1)
dispenser 4 level: 153 (stored: 1)
dispenser 5 level: 139 (stored: 1)
dispenser 6 level: 113 (stored: 2)
Checking levels done. New state: 0
Current state 6, event 0. No next state.