partyrobotics/bartendro #166: Disable 'Bartendro' Wifi and Login

gmuse, 2017-06-29:
How do I disable the hostapd/dnsmasq bartendro wifi and just access the UI through the IP my router gives the Pi? Accessing the (pop up) interface through the bartendro wifi is very inconsistent with my iphone, and doesn't work with my macbook. And accessing the interface through a browser doesn't work for either. 

I'd also like to disable the login, what's the easiest way to do this?

mayhem, 2017-07-04:
In /etc/rc2.d there should a file called S01hostapd (or similar, numbers might be different, I don't recall). Remove that file and reboot and the wifi access point feature should be disabled. While you're at it, remove the Wifi dongle from Bartendo for good measure.

gmuse, 2017-07-06:
@mayhem Thanks, that worked. Now the UI successfully comes up when I go to the local IP.

Is there an easy way to disable the login system so it goes straight to the admin UI? 

Also, how come I can't sudo apt-get install anything when ssh'd in? The password goes through but then I get a bunch of Err and Failed messages on the package installs.

mayhem, 2017-07-11:
As for the passwd, try changing the passwd to no password at all. I think that will skip the login.

sudo apt-get update

should update the package indexes so you can install more stuff.