partyrobotics/bartendro #167: Liquid level sensor

sdnavin, 2017-07-02:
Hi, 
I want to know how can i install the liquid level sensor and how the connection works , Please guide me on this.

pmich, 2017-07-05:
Please view instructions here:
http://support.partyrobotics.com/Guide/Installing+a+Liquid+Level+Sensor/11

sdnavin, 2017-07-06:
Hi @pmich This instructions is fine, We did this already but we have problems in that

The current Setup is we have two tubes going to the input drink, One from liquid level sensor and another from the dispenser IN. I believe this setup is fine, Now when i try setting the liquid level in the interface, the value i am getting - "4" all the time.

Debug log as follows :

Detected dispensers version 4. (Only checked first dispenser)
mixer.check_liquid_levels: check levels
dispenser 0 level: 4 (stored: 0)
Checking levels done. New state: 0
mixer.check_liquid_levels: check levels
dispenser 0 level: 4 (stored: 0)
Checking levels done. New state: 0
send_packet: Invalid ACK code 255
Failed to read liquid level threshold from dispenser 1

pmich, 2017-07-06:
The order of operations matters. You should slowly insert the dry tube (after it has already been attached to the liquid sensor), into the liquid, this is how you build positive pressure relative to ambient.

In the Liquid Sensor tab of the admin UI, you should be able to calibrate the value and see it on the spot, if you move the height of the tube relative to the liquid and you don't see calibration value changing, then there's a chance that the sensor is not plugged in all the way, or it is possibly damaged in some way. 

If you have a multi-meter, you can measure the output voltage on pin 1 (while it is plugged into the dispenser and powered) and see if the sensor is functional.