partyrobotics/bartendro #173: Master Password ?

NielsBar, 2023-06-30:
Hey,

 I hope you are well. I have a problem before Corona, I changed the password in the software and forgot it and unfortunately I didn't write it down. Is there a master password or can I somehow reset it? Thank you

mayhem, 2023-07-03:
If you can mount the SD card (which contains a linux partition) you and look at the sqlite database that stores the bartendro config and find the password there. Is that something you can do?

NielsBar, 2023-07-05:
Hi Robert, thank you very much, that worked. Do you still sell spare parts/machines or has everything been dissolved? I would need the boards for the pumps. How much would this cost and shipping to Germany at Umlostraße 94, 33649 Bielefeld?Thanks

mayhem, 2023-07-05:
On Wed, 2023-07-05 at 07:43 -0700, NielsBar wrote:
> Hi Robert, thank you very much, that worked. Do you still sell spare parts/machines or has everything been dissolved?

Everything has now been dissolved. 😞

-- 
--
--ruaok

Robert Kaye -- ***@***.*** -- metabrainz.org

NielsBar, 2023-07-05:
I'm sorry for you, could I maybe buy used or demonstration items from you that I somehow still come parts, or maybe board plans to recreate or something? Thanks so much

Everything has now been dissolved. 😞

pmich, 2026-03-16:
> Hi Sydney,
>
> Now that Rob is no longer with us, I may be the only person that can help
> you. I am the co-creator of Bartendro. I was the hardware brains and Rob
> was the software brains, so my ability to help you will be limited.
>
> The password I remember was: boozemeup
>
> I wouldn’t be able to tell you much about where to find the database files
> though.
>
> -Pierre

Username: bartendro