partyrobotics/bartendro #174: Comm's In J2 to RPI pins

Just-Tinkering, 2024-03-03:
Hi, I am trying to connect the pump board to the RPI through the RJ45 comm's jumper J2. I found the pin-outs for the pump [here:](https://cdn.sparkfun.com/datasheets/Widgets/Bartendro_Dispenser.pdf) and I'm using [this](https://github.com/partyrobotics/bartendro/releases/tag/v-2021-05-21) updated SD card image. 
However, after looking through the documentation I couldn't figure out where on my old RPI (modB-512M) does the TX, RX, Reset and Sync go.
I have come across a couple of [these](https://core-electronics.com.au/bartendro-dispenser-peristaltic-pump-and-controller.html) pumps and any help to get these pouring drinks would be greatly appreciated :)

mayhem, 2024-03-04:
TX and RX go to the serial port on the RPi that you want to use. (GPI14 and 15 for serial0). Sync can go to any GPIO pin -- the PRi should output 1ms square wave -- this is used to sync the LEDs in the pumps.

Just-Tinkering, 2024-03-05:
Thanx mayhem. 
Here is what I have done so far:
Loaded the above mention SD card image and inserted into the [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html)
I've inserted an RJ45 plug with cable into the dispenser and connected the other end as follows:
Pin 1 - 12v
Pin 2 - 5v
Pin 4 - Pin 8 (TxD UART) [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html)
Pin 5 - Pin 10 (RxD UART) [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html)
Pin 7 - Pin 12 (GPIO1) [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html)
Connected the [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html) to a monitor via HDMI.
I also added a Wi-Fi WiPi adapter to the RPI.

When I turn the power on the dispenser lights go blue.
The monitor displays green "OK's" for everything except for: 
[FAILED] Failed to start /etc/rc.local compatibility.
See 'systemct1 status rc-local.service' for details.

I can then log into the RPI directly via usb keyboard with: bartendro, hackme! but don't know what to type to test it.

I can log into it wireless with: bartendro, boosemeup but in MS Edge on win10 it says: 
Bartendro failed to startup, likely due to hardware or misconfiguration issue. Cannot open serial interface to router board.

Does that [SD-Card image](https://github.com/partyrobotics/bartendro/releases/tag/v-2021-05-21) work with my old [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html) on Win10 via MS Edge?
 
I'm sorry if these questions sound a bit silly but it's been a long time since I've done anything like this and we've found a few of these new in the box and would absolutely love to bring them to life.

Thanx again,
I really do appreciate your help.
:)

mayhem, 2024-03-05:
The rc.local failure relates to the fact that it can't find a real time clock. safe to ignore.

> Cannot open serial interface to router board.

Pin 4 - Pin 8 (TxD UART) [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html)
Pin 5 - Pin 10 (RxD UART) [RPI B (Rev1)](https://pi4j.com/1.2/pins/model-b-rev1.html)

I don't recall the pinout of the ethernet cable, but RX should connect to TX and vice versa. Double check those or just try swapping them and see if that improves it.

Just-Tinkering, 2024-03-08:
Hi mayhem,
I swapped the RX and TX but still no go :(
Do I need a router board if I'm only testing 1 dispenser?
Is there anything I can type into the RPI to test the dispenser when wired as above and connected to a HDMI monitor and USB keyboard?
Sorry for all the questions but I'm running out of ideas for this.

mayhem, 2024-03-11:
No router is required for the single pump, correct. Have you read this document?  https://github.com/partyrobotics/bartendro/blob/master/docs/dispenser.md

Just-Tinkering, 2024-03-12:
Yes I have. It says if the lights are blue it's awaiting a response. Type '?' to get a one byte response that is its dispenser id. 
The lights are blue but I can't get a response from '?'  I must admit I haven't researched how to send a 0xFF character or packet structures yet but I didn't think there would be much point if the ? character doesn't communicate. I was wondering if there was anything else I could type to test communications, if so could you quote it exactly please? Sorry if I sound a bit simple on this but I just want to make sure it's working before I get in too deep.
Thanx again for your time :)

mayhem, 2024-03-13:
OK, yes the first character should get a response, so its likely a serial comms issue. Have you enabled the serial port using raspi-config ? Do you have a logic analyzer to see what is being sent over the lines?

Just-Tinkering, 2024-03-20:
>  Maybe try using a more recent distro -- perhaps something about the serial port setup has changed in the meantime?

This is why I am using an old RPI (modB-512M) and asked if it was compatible with your image?
Which RPI would you recommend for your image?
What do you mean 'more recent distro'. Isn't that your image?
Is the '?' is typed from the keyboard connected directly to the RPI?
Sorry if my questions sound silly but I want to make sure I'm not missing anything, links and examples are greatly appreciated.
Thanx.

Just-Tinkering, 2024-03-29:
Hi Mayhem, hopefully I haven't annoyed you too much. My aim for this issue was to get foolproof instructions to set this up so someone who is not familiar with RPI's etc can get this working. For example, when you said 'try using a more recent distro', do I have to install that then install your image? I thought your image included an OS but like I said, I need 'foolproof' instructions. 
I realize it was a long time ago but it seems like too good of a project to let slip away.
Much appreciated,
Just-Tinkering.

mayhem, 2024-03-29:
Sorry, life is busy right now. What I need to do is sit down with a pump and get it to work actually. For that i need a couple of quiet hours, which have been hard to come by... Next week hopefully.

Just-Tinkering, 2024-03-29:
No need to be sorry, I am extremely grateful for any help you are willing to give especially since this is a very old project. I was actually logging on to delete my last comment because I thought I was being too pushy. Take as long as you want, I absolutely agree with the busy life statement.
Thanx again,
Just-Tinkering.

mayhem, 2024-04-03:
OK, I connected a pump directly to a Raspberry Pi and I was able to  communicate with the pump. To get the connection right here are the details, looked up to be sure:

[image]

The schematic shows you what the wires want and the RJ45 header shows you which connection goes where:

Brown (GND) -> pin 6 (GND) (board not gpio)
Blue white (TX) -> pin 10 (RX)
Blue (RX) -> pin 8 (TX)
Orange (5V) -> pin 2

the connect with "pyserial-miniterm  /dev/serial0" which defaults to 9600 baud, which is correct. I type ? and immediately get a response of '2'. Your response would likely differ.

Key thing to remember, even though your are feeding 5V into the dispenser, the mcu on the dispenser runs at 3.3V, so the IO pins must use 3.3V, NOT 5.0V.

IHTH.